arduino-libraries/Arduino_SensorKit #8: Consider U8g2 instead of U8x8?

marcmerlin, 2020-12-17:
Howdy. I got an eval unit from Elaine Wu and sent her a bunch of feedback on issues i thought made this harder to use than it should, for complete beginners.
The documentation on OLED is somewhat lacking, for one it mentions nothing about https://github.com/olikraus/u8g2
U8g2
Includes all graphics procedures (line/box/circle draw).
Supports many fonts. (Almost) no restriction on the font height.
Requires some memory in the microcontroller to render the display.

U8x8
Text output only (character) device.
Only fonts allowed with fit into a 8x8 pixel grid.
Writes directly to the display. No buffer in the microcontroller required.

For now, you have chosen U8x8, but it contains some serious limitations. You should explain how it can be used, and not used. it's non obvious enough that your own oled example used the library wrong (as per https://github.com/arduino-libraries/Arduino_SensorKit/pull/7 )

If the intent is to save memory on those underpowered and obsolete Unos that only have 2KB of RAM (although U8g2 works on them), this should be explained, along with the fact that Oled does not allow drawing lines or anything since it uses U8x8 and not U8g2

marcmerlin, 2020-12-18:
Thanks for confirming the 2KB memory problem I was afraid was the cause for this.
I wasn't going to spam the bug with tangential stuff, but I noticed your board has a 3.3/5V switch which isn't really documented (i.e. do you fry the MCU if the switch is on 5V and the MCU is 3.3V?), but if it's used properly, your board should work fine with a bunch of boards that have so much more memory, be it arduino 101, or your own https://www.seeedstudio.com/Seeeduino-Lotus-Cortex-M0-p-2896.html

Your cortex board had 32KB for half the price of an official arduino Uno with 2KB or barely more than your own seeeduino 4.2: https://www.seeedstudio.com/Seeeduino-V4-2-p-2517.html
The Uno is now 10 years old, completely obsolete, and totally inadequate for many tasks due to the lack of RAM. I think it would serve your users to introduce them to arduino compatible sensors and programming on a board that isn't so obsolete and limited.

marqdevx, 2020-12-18:
You can switch the 3V3/5V switch, it will not make any damage to the Arduino Board cause it has its own regulator.
The only issue is that the Display only works with 5V.

I will add the u8g2 example next week.

marcmerlin, 2020-12-18:
Thanks. I moved the 3.3/5V discussion to https://github.com/arduino-libraries/Arduino_SensorKit/issues/10 to avoid crossing streams :)

marcmerlin, 2021-01-12:
In the meantime, I filed https://github.com/arduino-libraries/Arduino_SensorKit/pull/12 to make it more clear that this library is using U8x8

closed 2021-06-28 by marqdevx