arduino-libraries/Arduino_SensorKit #9: Examples using I2C should explain that all I2C devices on the board conflict

marcmerlin, 2020-12-17:
Unfortunately they were not configured to use different addresses (although it seems possible with small solder points).
An obvious use of the kit is to read pressure or some other I2c value (like accelerometer), but that's not possible since they share I2C and conflict with the OLED. If you turn them on, the OLED stops working.
Not obvious at all for a beginner, documentation is needed.

marqdevx, 2020-12-17:
Hey!
We are using u8x8 to save memory, if we use u8g2 we use more than 80% of the memory.

This kit is meant to use it with the content so maki the things simple was te main goal, but you are right that it has some graphic limitations, noted.

I will add some examples using the u8g2 library for advanced users that want to try more capabilities of the HarWare, thanks a lot for your feedback!!

Have a nice one.

per1234, 2020-12-18:
@marqdevx I believe your reply was intended to be posted to the other issue: https://github.com/arduino-libraries/Arduino_SensorKit/issues/8

Since this issue has not been addressed, I'll reopen.

marcmerlin, 2020-12-18:
Thanks for confirming the 2KB memory problem I was afraid was the cause for this.
I wasn't going to spam the bug with tangential stuff, but I noticed your board has a 3.3/5V switch which isn't really documented (i.e. do you fry the MCU if the switch is on 5V and the MCU is 3.3V?), but if it's used properly, your board should work fine with a bunch of boards that have so much more memory, be it arduino 101, or your own https://www.seeedstudio.com/Seeeduino-Lotus-Cortex-M0-p-2896.html

Your cortex board had 32KB for half the price of an Uno with 2KB: https://www.seeedstudio.com/Seeeduino-Lotus-Cortex-M0-p-2896.html
The Uno is now 10 years old, completely obsolete, and totally inadequate for many tasks due to the lack of RAM. I think it would serve your users to introduce them to arduino compatible sensors and programming on a board that isn't so obsolete and limited.

marcmerlin, 2020-12-18:
I tried to give a better combined example that exercises most sensors, but wasn't able to combined them all due to this problem. The partial example is here
https://github.com/arduino-libraries/Arduino_SensorKit/pull/6

I'm pretty sure I can make it work by soldering address points on the sensors, but then I'd have code that only works on my board, so it's not useful as an example to your users.

marqdevx, 2020-12-18:
> 
> 
> @marqdevx I believe your reply was intended to be posted to the other issue: #8
> 
> Since this issue has not been addressed, I'll reopen.

Oops, my fault!

marqdevx, 2020-12-18:
Hey @marcmerlin
Could you please run the i2c scanner individually for each sensor? and give me your i2c addresses.

I can run both at the same time.

marcmerlin, 2020-12-18:
My bad, you are correct.
```
I2C device found at address 0x19  !
I2C device found at address 0x3C  !
I2C device found at address 0x77  !
```

Now I need to find out why initializing 
//Accelerometer.begin();
or
 //Pressure.begin();

stops the OLED from working.
It's been a while since I've worked with I2C, do the drivers auto-scan all addresses to find the right device, and maybe send wrong init messages to the OLED that mess it up?

marcmerlin, 2020-12-18:
For what it's worth, running the I2C scanner first, also stops the OLED from working, but that might be a different issue.
Either way, it'd be great to have a generic example, like the one I submitted, that exercises everything, proves that it all works, and helps a novice user running all of them correctly?

marqdevx, 2020-12-21:
I'll add an example
thanks for the feedback!! @marcmerlin

noppatoppa, 2021-01-04:
> An obvious use of the kit is to read pressure or some other I2c value (like accelerometer), but that's not possible since they share I2C and conflict with the OLED. If you turn them on, the OLED stops working.

Can confirm this: using air pressure sensor disables OLED.

marqdevx, 2021-01-12:
Hello @marcmerlin @noppatoppa 
First, thanks for the feedback
And second, could you please test this branch and let me know if now its working fine?
https://github.com/arduino-libraries/Arduino_SensorKit/tree/display_hw_i2c

I think it has been some update in the display library, cause the SW I2C for the display was fixing an issue that we had, but now the issue came back, and switching back to the HW I2C seems to fix it.

Let me know
Have a good one! :D

marcmerlin, 2021-01-12:
I can confirm, this works, thank you.
I sent you a PR of your own code :)  https://github.com/arduino-libraries/Arduino_SensorKit/pull/13
And if you don't mind, I'd be honored if you included my full example that uses all the sensors on the board
https://github.com/arduino-libraries/Arduino_SensorKit/pull/6

[image]

marqdevx, 2021-01-13:
I will have a look, I will talk in the Pull Request

closed 2021-01-13 by marqdevx