arduino-libraries/Arduino_SensorKit #10: Documentation update; 3.3/5V switch on the board

marcmerlin, 2020-12-18:
@marqdevx , I'll open a separate issue for this.
As a reminder, I'm not asking for personal help, or being demanding :) , I was given this board to test and give feedback on, pretending that I'm a new user, so I'm doing just that :)

You said "You can switch the 3V3/5V switch, it will not make any damage to the Arduino Board cause it has its own regulator. The only issue is that the Display only works with 5V."

So, the arduino board, even if it has a 3.3V MCU, has a 5V line. Therefore your board receives 5V in both cases, but on the VREF line, received 3.3V or 5V depending on what arduino you have (the newer/better ones, all being 3.3V).
The issue is that if any sensor sends back 5V on a data line that is a 3.3V only GPIO on the board, that GPIO may get fried (you know this of course, just writing it out to be clear)

Documentation that explains what the switch does, would help, does it have a 2nd regulator on the board to take 5V back to 3.3V? If so, why not just take the 3.3V from the arduino?

Some devices (you said the OLED) only work with 5V data input? (so a 3.3V high from an arduino 101 will not be able to talk to it?)

Basically a page explaining all this in your documentation would be welcome, I know some about those issues, but I'm not fully clear what the switch does, and what devices won't work at 3.3V and why? (it's typically for 5V devices to also work with 3.3V, even neopixels typically work with 3.3V signals)

marcmerlin, 2020-12-21:
Thinking about this some more, I think I know what you're doing:
All sensors, except maybe one? Work with 3.3V or 5V.
The switch decides whether you are taking the 3.3V (Vref) instead of 5V (Vusb) from the arduino. On a 3.3V arduino, you definitely want to feed 3.3V to the sensors so that they don't try to send back 5V logic to the MCU's GPIO pins, which could fry them.
The part where I'm confused is: why don't you take the Vref pin from the arduino which will be 3.3V or 5V depending on the switch and do the right thing no matter what? If you did that, you wouldn't need the switch at all?

marqdevx, 2021-01-11:
In this case there is no issue with the 5V and the 3V3 GPIOs.

The carrier board is not only made for this kit.
We cannot remove that switch.

Thanks for your feedback! :) 
Have a good one

marqdevx, 2021-01-11:
Btw I will try to add some explanation about the switch issue in the content, I forgot to tell you.
Thanks again!

closed 2021-01-11 by marqdevx

marcmerlin, 2021-01-12:
Hi  @marqdevx , thanks for your reply.
Just so that we're on the same page:
- I wasn't asking you to remove the switch, just to document how it can be used
- Yes, I understand that with a 5V uno or mega it's not needed, but for a 32bit 3.3V CPU, it probably is. As per my previous questions, I'm not entirely sure how it helps, though
- thanks for updating the documentatoin. Can you give me a link to the updated page I should read?

marcmerlin, 2021-03-04:
hi, any update on that point?