arduino-libraries/Arduino_SensorKit #15: Typo in U8X8_SSD1306_128X64_NONAME_HW_I2C Oled constructor?

ppk42, 2021-01-25:
I found the following code from the U8g2 Lib. 
```
class U8X8_SSD1306_128X64_NONAME_HW_I2C : public U8X8 {
  public: U8X8_SSD1306_128X64_NONAME_HW_I2C(uint8_t reset = U8X8_PIN_NONE, 
             uint8_t clock = U8X8_PIN_NONE, uint8_t data = U8X8_PIN_NONE) : U8X8() {
    u8x8_Setup(getU8x8(), u8x8_d_ssd1306_128x64_noname, 
             u8x8_cad_ssd13xx_fast_i2c, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_HW_I2C(getU8x8(), reset, clock, data);
  }
};
```
The order of the contructor parameters are: reset, clock, data.

In Arduino_SensorKit.cpp An instance (Oled) is created with a different signature.

`U8X8_SSD1306_128X64_NONAME_HW_I2C Oled(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);`

Did I miss something, or is this an error? if it is an error, should I fork and contribute?

marqdevx, 2021-01-25:
Hey!
You are right, it shoud have only the reset inside the parameters
[](https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#ssd1306-128x64_noname-1)

I will try tomorrow, I dont have that display in my hands.
Maybe its doing SW I2C, so I need to check.

But if you can try by changing the constructor to 

> U8X8_SSD1306_128X64_NONAME_HW_I2C Oled(U8X8_PIN_NONE);

Theoricaly it should work.

Thanks! Have a nice one :)

ppk42, 2021-01-26:
Hi!
 
Sorry, I was very bussy, so I'm a little bit late. ;-) 

I found this issue when, I was answering a stack overflow question: [SO Post - Grove-Base-Kit multiple I2C connections](https://arduino.stackexchange.com/questions/80875/grove-base-kit-multiple-i2c-connection)

I have no access to the Starter kit, so I can not try out. I'll ask on Stack overflow if the OP could check it.
As the constructor parameters all have a default of `U8X8_PIN_NONE`, I assume, we could use an empty constructor, like:

`U8X8_SSD1306_128X64_NONAME_HW_I2C Oled;`

If you have no caveats, I'll ask fireblazer10 on SO to gently try that.  

I saw in the revision history that the I2C of the Oled was changed from HW to SW, then back to HW. It looks like the signature was not changed with the last switch.

marqdevx, 2021-01-26:
I've tested it and it seems to work, I will make a new release.

I will have a look at the post anyway in case I can solve some questions.

Thanks!!
Have a nice one.

closed 2021-01-26 by marqdevx