arduino-libraries/Arduino_SensorKit #21: OLED Display not working with Air Pressure Module on I2C

Xiryl, 2021-04-30:
Hi there.

I am doing some testing on the Arduino sensor kit base. Specifically, I connected the "Air Pressure" sensor and the "OLED Display" to the support board, both to the I2C connectors.

Using them alone, the air pressure sensor, or the OLED display, work perfectly. However, when I try to use them together, the OLED display stops working.

I tried downloading the latest version of the library (`1.0.9`) and also importing it as a `.zip` from this repo.

I conclude that if I try to use the `Pressure.begin();` instruction, the OLED display stops working.
How I can resolve this?

Thanks in advice!

I Attach my sample code.
```C
#include <Arduino_SensorKit.h>

void setup() {
  Serial.begin(9600);

  Environment.begin();

  Oled.begin();
  Oled.setFlipMode(true);

  Pressure.begin(); // <- display not working using this line
}

void loop() {
  Oled.setFont(u8x8_font_amstrad_cpc_extended_r); 
  Oled.setCursor(0, 0);
  Oled.print("Hello world"); 
  delay(100);
}
```

per1234, 2021-04-30:
Here was a previous report here:
https://github.com/arduino-libraries/Arduino_SensorKit/issues/9#issuecomment-754139149
But that was reported as having been fixed.

Xiryl, 2021-04-30:
Thanks @per1234 .

I tried now to test it on the mkr 1010 wifi and it works perfectly, but with the arduino one R3 still not..

marqdevx, 2021-05-01:
Hey! Sorry for te issue.

Could you try the library version 1.0.8 please? Maybe it should fix it for you.

Anyway i will investigate more deep, the problem is in the Display's library.

I will let you know as soon as possible!

marqdevx, 2021-05-01:
There is a issue with the display's library, i will investigate how to solve it.

I will let you know asap!

UnfoldAntwerp, 2021-05-02:
> Here was a previous report here:
> [#9 (comment)](https://github.com/arduino-libraries/Arduino_SensorKit/issues/9#issuecomment-754139149)
> But that was reported as having been fixed.

Landed here after endless searching why the OLED stops working if I use the other I2C sensors on my Arduino Sensor Kit. Looks like the latest version 1.9 reverts back to Software for OLED I2C. I edited the .h & .cpp files to revert the SW back to HW and now it works!

Looks related to this issue: https://forum.seeedstudio.com/t/i2c-module-problems-with-grove-beginner-kit-for-arduino/252416/28. Will check what happens if I remove the module from the frame at a later point. Mine are still attached for now.

marqdevx, 2021-05-03:
@Xiryl could you please add `Oled.setBusClock(100000);` before `Oled.begin()` and let me know if it solves it? if it doesnt try to download the v1.0.8 and try the same.

@UnfoldAntwerp I reverted cause we have some issues with the Display aswell, and it was not working with the HW I2C, could you let me know wich board are you using?

Thanks!!

marqdevx, 2021-05-03:
It seems that the issue is generated on the BMP280 sensor, im on it, I will update you ASAP

UnfoldAntwerp, 2021-05-04:
@marqdevx Arduino Uno SMD R3

Xiryl, 2021-05-06:
@marqdevx unfortunately I have been using the display on the mkr 1010, so I can't do the test with the arduino one at the moment

kootsoop, 2021-06-25:
If it helps, adding `Oled.setBusClock(100000);` before the `Oled.begin();` doesn't help for me in Xiryl's code.

marqdevx, 2021-06-28:
Thanks for the info @kootsoop I'm still running some tests (I will try that) :)

gcreate42, 2021-07-01:
> If it helps, adding `Oled.setBusClock(100000);` before the `Oled.begin();` doesn't help for me in Xiryl's code on an Uno R3.
> 
> I reverted to version `1.0.8` of the library, and everything worked again using the out-of-the-box `Combined_Demo` code.

Reverting to 1.0.8 fixed this problem for me as well. I had the same problem with the Combined Demo which was frustrating right out of the box.

marqdevx, 2021-07-05:
Hey @Xiryl @gcreate42 Could you try to run individually the Pressure sensor and the Display? (Using v1.0.8)
Thanks! Have a nice one!

Arduinux, 2021-08-24:
Also for me, going back to 1.0.8 fixes it.

marqdevx, 2021-08-24:
Could you please share the board that you are using @Arduinux ?
Thanks for your feedback. :)

Arduinux, 2021-08-24:
It's an Arduino Rev. 3 with the golden lying eight on the fuse.

Bodobolero, 2022-02-16:
@marqdevx It is a little annoying that a bug that was reported so long ago and that still worked reliably in 1.08 (so it is a regression reported by multiple users) is still not fixed in current version.
You could save people a lot of time by reverting the bug introduced in 1.09

Thanks for your understanding

sp-rky, 2022-04-20:
Hello, 

After having this issue, I tried both downgrading the library to 1.0.8 and changing to hardware in Arduino_SensorKit.h and Arduino_SensorKit.cpp, however my display refused to work, even after disabling all other I2C devices. 

What fixed it for me was commenting out `Wire.setClock(u8x8->bus_clock);` on line 1345 in U8x8lib.cpp in the U8g2 library. This took a few hours to figure out, so I hope this helps someone out :)

slavicd, 2022-10-13:
Erm, guys? It's 2022. This is not reported anywhere in docs and I really could have spent less than all these hours wondering what I'm doing wrong.

marqdevx, 2022-10-17:
This should be fixed from v2.33 of the library then, am I wrong? 🤔 
@sp-rky @Bodobolero

TomB4321, 2023-01-18:
Arduino Sensor Kit library version 1.0.9 doesn't seem to work with the OLDE screen, I've reverted to version 1.0.8 and my screen in working with all the sensors ont he board again!

MiBiCZ, 2023-09-09:
It's very sad, but the bug is still not fixed even in version 1.0.10. What can be done to finally get someone to do it? :-( Thank you. MiBi

alranel, 2023-12-15:
This is fixed as of 1.4.0 (which also adds compatibility with UNO R4 Minima, UNO R4 WiFI, GIGA R1 WiFi).

closed 2023-12-15 by alranel