arduino-libraries/Arduino_SensorKit #31: Temperature & Humidity Sensor #define DHTPIN does not work.

andrew-rock, 2022-04-20:
The library works fine for pin 3. The method for assigning another digital pin using a #define DHTPIN yourPin does not work and probably can't be made to work. It does not work even for the example file provided with the library, Temp_and_Humidity.ino.
The problem is that Temp_and_Humidity.ino and Arduino_SensorKit.cpp are compiled independently. The compiler sees the #define DHTPIN yourPin and the does not redefine it to 3 when compiling Temp_and_Humidity.ino, but it can not see it when compiling Arduino_SensorKit.cpp, so the Environment object as DHT(3, DHT11). The same problem will happen with trying to change the DHTTYPE with a #define.

per1234, 2022-04-21:
Thanks @andrew-rock. I confirm that it is impossible to configure the pin as claimed via the sketch due to it being a separate [translation unit](https://en.wikipedia.org/wiki/Translation_unit_(programming)) from `Arduino_SensorKit.cpp`.

I see this incorrect information has been stated in several places:

- https://github.com/arduino-libraries/Arduino_SensorKit/blob/main/examples/Temp_and_Humidity/Temp_and_Humidity.ino
- https://github.com/arduino-libraries/Arduino_SensorKit/blob/main/docs/readme.md
- https://sensorkit.arduino.cc/sensorkit/module/lessons/lesson/08-the-temperature-sensor

marqdevx, 2022-08-25:
Hello!
i've run this test:

sketch:
```cpp
#define TEST 1
#include "test.h"

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while(!Serial);
  Serial.println(TEST);
 
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

test.h
```cpp
#ifndef TEST
  #define TEST 2
#endif
```

The output if I define it on the sketch its 1, if i dont define it is 2. Im missing something 🤔
Is the problem when the constructor is being compiled?

per1234, 2022-08-25:
>Im missing something 🤔

Yes. The subject of translation units can be quite confusing.

The reason why it works as expected in your test sketch is because there is only one translation unit: the `.ino` file of the sketch.

You can think of an `#include` directive as being replaced by a copy/paste of the contents of the specified file, so the code from `test.h` becomes part of the `.ino` file's translation unit and all is well.

The problem occurs when you try to use a macro defined in the scope of one translation unit in another translation unit.

Try this sketch, which contains two translation units (`SomeSketch.ino` and `test.cpp`):

**SomeSketch.ino**

```cpp
#define TEST 1
#include "test.h"

void setup() {
  Serial.begin(9600);
  while(!Serial);

  Serial.print("Value of TEST in .ino translation unit: ");
  Serial.println(TEST);
 
  printMacroValue();
}

void loop() {}
```

**test.h**

```cpp
#ifndef TEST
  #define TEST 2
#endif

void printMacroValue();
```

**test.cpp**

```cpp
#include <Arduino.h>
#include "test.h"

void printMacroValue() {
  Serial.print("Value of TEST in test.cpp translation unit: ");
  Serial.println(TEST);
}
```

alranel, 2023-11-07:
This is now fixed in 1.3.0 which implements a new `Environment.setPin()` method to be called before `begin()`:

```
void setup() {
  //uncomment line below if you're connecting your DHT20 to pin a different than 3
  //Environment.setPin(4);

  Serial.begin(9600);
  Environment.begin();
}
```

Documentation will be fixed as soon as possible.

alranel, 2023-11-09:
Documentation fixed as well!

closed 2023-11-09 by alranel