arduino-libraries/Arduino_SensorKit #32: Compatibility of Sensor kit library with Arduino Nano RP2040 connect

Bodobolero, 2022-04-28:
When I compile using the Sensorkit for mbed_nano architecture (Arduino Rp2040 Connect board) I get the following warning and error messages:

```
library Arduino_Sensorkit claims to run on avr architecture(s) and may be incompatible with your current board which runs on mbed_nano architecture(s)
...
Compiling library "Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR"
/Users/peterbendel/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/variants/NANO_RP2040_CONNECT/defines.txt @/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -MMD -mcpu=cortex-m0plus -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/variants/NANO_RP2040_CONNECT -I/Users/peterbendel/Documents/Arduino/libraries/Arduino_Sensorkit/src -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/libraries/Wire -I/Users/peterbendel/Documents/Arduino/libraries/Grove_-_Barometer_Sensor_BMP280 -I/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/libraries/SPI -I/Users/peterbendel/Documents/Arduino/libraries/Grove_Temperature_And_Humidity_Sensor-master -I/Users/peterbendel/Documents/Arduino/libraries/U8g2/src -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/api/deprecated -I/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/api/deprecated-avr-comp -iprefix/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino @/Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/variants/NANO_RP2040_CONNECT/includes.txt /Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp -o /private/var/folders/97/6vhtgg3s5z3frxr6g2c4x8s00000gn/T/arduino-sketch-CDAB3820DEAA2C5841572D5994559EDF/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/LIS3DHTR.cpp.o
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp: In member function 'void LIS3DHTR<T>::begin(arduino::SPIClass&, uint8_t)':
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp:49:15: error: 'arduino::SPIClass {aka class arduino::HardwareSPI}' has no member named 'setClockDivider'
     _spi_com->setClockDivider(SPI_CLOCK_DIV4);
               ^~~~~~~~~~~~~~~
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp:49:31: error: 'SPI_CLOCK_DIV4' was not declared in this scope
     _spi_com->setClockDivider(SPI_CLOCK_DIV4);
                               ^~~~~~~~~~~~~~
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp:49:31: note: suggested alternative: 'SPI_CPOL_1'
     _spi_com->setClockDivider(SPI_CLOCK_DIV4);
                               ^~~~~~~~~~~~~~
                               SPI_CPOL_1
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp:51:15: error: 'arduino::SPIClass {aka class arduino::HardwareSPI}' has no member named 'setBitOrder'
     _spi_com->setBitOrder(MSBFIRST);
               ^~~~~~~~~~~
/Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR/src/LIS3DHTR.cpp:55:15: error: 'arduino::SPIClass {aka class arduino::HardwareSPI}' has no member named 'setDataMode'
     _spi_com->setDataMode(SPI_MODE3);
               ^~~~~~~~~~~

Using library Arduino_Sensorkit at version 1.0.8 in folder: /Users/peterbendel/Documents/Arduino/libraries/Arduino_Sensorkit 
Using library Wire in folder: /Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/libraries/Wire (legacy)
Using library Grove_-_Barometer_Sensor_BMP280 at version 1.0.1 in folder: /Users/peterbendel/Documents/Arduino/libraries/Grove_-_Barometer_Sensor_BMP280 
Using library Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR at version 1.2.3 in folder: /Users/peterbendel/Documents/Arduino/libraries/Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR 
Using library SPI in folder: /Users/peterbendel/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/libraries/SPI (legacy)
Using library Grove_Temperature_And_Humidity_Sensor-master at version 1.0.0 in folder: /Users/peterbendel/Documents/Arduino/libraries/Grove_Temperature_And_Humidity_Sensor-master 
Using library U8g2 at version 2.31.2 in folder: /Users/peterbendel/Documents/Arduino/libraries/U8g2 
Compilation error: exit status 1
```

facchinm, 2022-04-28:
Hi @Bodobolero ,
the issue is in a dependency library ( Grove-3-Axis-Digital-Accelerometer-2g-to-16g-LIS3DHTR ) which was updated to use SPISettings instead than single calls https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/commit/c56da9d58d5477ac14efd190d05579f797d2edb8 but was not released in the meantime.
Updating it manually with the code from github https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/archive/refs/heads/master.zip should fix the issue.

closed 2022-04-28 by facchinm

Bodobolero, 2022-04-29:
The LIS3DHTR library was not released sine Nov 20.
https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/releases/tag/v1.2.3

Does the Arduino org have any influence on Seeed release cycle to avoid people running into this problem and wasting a lot of time figuring out the problem ?

per1234, 2022-04-29:
@Bodobolero please open a request on the library's issue tracker:

https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/issues

My experience when contributing to several Seeed Studio repositories over the years has been that the maintainers are commendably responsive.

Bodobolero, 2022-04-29:
> @Bodobolero please open a request on the library's issue tracker:
> 
> https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/issues
> 
> My experience when contributing to several Seeed Studio repositories over the years has been that the maintainers are commendably responsive.

Done, see https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR/issues/6